bbcmicrobit/micropython #669: micropython radio channel and MakeCode radio Group

simonc8, 2019-09-24:
I'm confused about how Radio Group in MakeCode relates to Radio Channel in micropython as I've failed to get any communication between one microbit programmed in micropython and another programmed in MakeCode, especially as micropython also has a setting for Group.

I'd appreciate if someone could clarify this.

ukBaz, 2019-09-24:
@simonc8 ,

MakeCode adds a little extra information in the radio packets.
I've written this up at: https://ukbaz.github.io/howto/ubit_radio.html

@rhubarbdog has written a class to help with communicating with the MakeCode Radio format:
https://github.com/rhubarbdog/microbit-radio

simonc8, 2019-09-25:
Thanks for these links. I was never going to get it to work simply exporting a string from micropython and expecting the MakeCode microbit to understand it. 
Still trying to make it work using make_radio, but at the moment I can't get the Mu editor to import make_radio from the same folder. It says "No module named make_radio".

ukBaz, 2019-09-25:
Your original issue has been answered so I'm not sure this the right place to continue this conversation.

You have not given enough information for me to guess what might be going wrong. I am assuming you are copying over the make_radio.py file to the micro:bit like shown on this page:
https://codewith.mu/en/tutorials/1.0/microbit

Another option is to copy and paste the make_radio class in to your main file.

There is also more information about modules at:
https://forum.micropython.org/viewtopic.php?t=4431

rhubarbdog, 2022-11-07:
You need to use file transfer to copy `make_radio.py` to your microbit