bbcmicrobit/micropython #675: OnBluetoothDisconnected event in radio module

niutech, 2020-03-30:
MakeCode for Micro:bit supports the [OnBluetoothDisconnected](https://makecode.microbit.org/reference/bluetooth/on-bluetooth-disconnected) event. Can the same be supported in [Radio](https://microbit-micropython.readthedocs.io/en/latest/radio.html) module, e.g. `radio.on_disconnected(callback)`?
Thanks!

kopp, 2020-03-30:
as far as I understand, the `radio` module does not have the concept of _connected_ or _disconnected_, but just of _on_ and _off_.

As far as I understand from [the code](https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/modradio.cpp) and [the discussion in the pull request introducing the feature](https://github.com/bbcmicrobit/micropython/pull/283) MakeCode and MicroPython use different software layers to get Radio working (MakeCode uses the DAL while MicroPython does not), so I would expect different features to be available in those two...

2bndy5, 2021-08-07:
The radio module doesn't use a pairing/bonding mechanism like Bluetooth does. Additionally, I don't see the underlying implementation using an auto-ACK feature, so there is no way to know if there is a connection at all. If there was an auto-ACK feature, it would negate using the term "group", since 1 transmission could be acknowledged by only 1 receiver.

Simply put, this request cannot be satisfied in any way.