bbcmicrobit/micropython #679: Search results in 500 error

microbit-mark, 2020-06-15:
Attempting to search readthedocs https://microbit-micropython.readthedocs.io/en/v1.0.1/ results in 500 nginx error or 502 bad gateway cloudflare error.

clicking on project home also has the same result
[image]
Links to
https://readthedocs.org/projects/microbit-micropython/?fromdocs=microbit-micropython

Firefox 77 on Mac 10.15
Tried v1.0.1 and latest docs

microbit-mark, 2020-06-15:
Now appears not to reproduce

microbit-mark, 2020-06-15:
Seems to have been fixed https://readthedocs.org/ doesn't seem to offer an indication of the status. They say that hosting is provided by Azure.

I'll feedback about having some kind of status page, but closing this ticket

closed 2020-06-15 by microbit-mark

microbit-mark, 2020-06-15:
https://github.com/readthedocs/readthedocs.org/issues/7192

microbit-carlos, 2020-06-15:
Thanks for keeping an eye on this and the feature request Mark!