bbcmicrobit/micropython #681: radio.config queue parameter

microbit-josephine, 2020-07-30:
The queue parameter in radio.config() does not seem to have an affect. Even when not set (and so the queue default is set to 3) the micro:bit seems to be able to store at least 10 messages in the queue.

# To recreate:
flash the contents of this script onto a micro:bit:
```
from microbit import *
import radio

# turn on the radio
radio.on()
cpt = 0

while True:
    if button_a.was_pressed():
        radio.send(str(cpt))
        cpt = cpt + 1

    sleep(100)
```

Then, by pressing A 5 times, send 5 different messages to another micro:bit set up to receive them - you will need to put into the terminal:

```
>>>from micro:bit import *
>>>import radio
>>>radio.config(queue=1)
>>>radio.receive()
0
>>>radio.receive()
1
>>>radio.receive()
2
etc
```

you should get at least 5 different messages, which shows more than 1 message is being stored on the queue.

dpgeorge, 2020-07-30:
I think what's happening here is that the radio buffer is allocated memory to store `length * queue` bytes, ie if you configure it to have messages up to 64 bytes and want to receive 5 such messages it allocates `64*5=320` bytes for the incoming buffer.  But if the messages are short then it will be able to store more, eg if each message is only 10 bytes then it can store 32 of them.

mytechnotalent, 2020-09-30:
Ask this question in the next generation of micro:bit for microcontrollers which is the MiniPython programming language.  Click HERE and re-ask your question. - https://github.com/mytechnotalent/minipython today!