bbcmicrobit/micropython #682: i2c device memory operations

ozanoner, 2020-08-28:
Hi, 
Is there any plan to add i2c device memory operations? as here:
[https://docs.micropython.org/en/latest/library/machine.I2C.html#machine.I2C.readfrom_mem](https://docs.micropython.org/en/latest/library/machine.I2C.html#machine.I2C.readfrom_mem)

I tried to use BME280 however the driver requires access to its internal registers. one example here:
[https://randomnerdtutorials.com/micropython-bme280-esp32-esp8266/](https://randomnerdtutorials.com/micropython-bme280-esp32-esp8266/)

Regards,
Ozan

dpgeorge, 2020-08-29:
No there is no plan to add those operations.

But it's possible to do memory operations with the existing 2 methods, eg:
```python
def read_mem(i2c, addr, mem_addr):
    i2c.write(addr, bytes([mem_addr]), True)
    return i2c.read(addr, 1)[0]
```

jemerlia, 2020-08-29:
Hi
Microbit works well with the BME280 but the computationally heavy code takes much of its memory.  Hope this snippet written for the BME280 helps:
# Discover if the sensor is there
# IDAddress     = bytearray(1)
# IDAddress[0]  = 0xD0# i2c.write(0x77, IDAddress, repeat = False)
# id = i2c.read(0x77, 1, repeat = False)
# print(id)

Best wishes => Terry

 
 

https://docs.micropython.org/en/latest/library/machine.I2C.html#machine.I2C.readfrom_mem

https://randomnerdtutorials.com/micropython-bme280-esp32-esp8266/

mytechnotalent, 2020-09-30:
Ask this question in the next generation of micro:bit for microcontrollers which is the MiniPython programming language.  Click HERE and re-ask your question. - https://github.com/mytechnotalent/minipython today!

paulsijben, 2021-01-18:
I am running into serious problems trying to get an MPU6500 to work for similar reasons. On a raspbery pi it works immediately, but trying to get this chip going on the microbit with the web-based python editor I see that for starters I need to do +1 on all the writes writes for the register number,  only then I can read the who_am_i register correctly. Trying to read the gyro and accel registers gives me data that suggests that the registers are somehow swapped. 

Is there any point in trying to checkout this repository and run the updates manually?