bbcmicrobit/micropython #683: deque class in ucollections module

victorzaloga, 2020-09-05:
Hello microbit society!
I`d like to code moving average filter by using of ucollections.deque class

but when trying to import the module got the error
**from ucollections import deque** -> "cannot import name deque" 
Is deque available in the module ucollections in microbit micropython?

mytechnotalent, 2020-09-30:
Ask this question in the next generation of micro:bit for microcontrollers which is the MiniPython programming language.  Click HERE and re-ask your question. - https://github.com/mytechnotalent/minipython today!

ZanderBrown, 2020-09-30:
> generation of micro:bit for microcontrollers

You mean micropython? also HERE isn't a link (other than the person linking to a fork on all the issues who are you? the fork just seems to rename micropython to minipython)

mytechnotalent, 2020-09-30:
Yes it is MicroPython but as it is built it will include steps to build your own modules and continue the development.  The ultimate goal is to get more people involved with MicroPython nothing more.

victorzaloga, 2020-09-30:
question about ucollections / deque is still unclear
Anybody could help me please?

mytechnotalent, 2020-09-30:
Hi @victorzaloga there is no deque within that module for the BBC firmware.  Here is how you can see what methods are available to you in the REPL.
`import ucollections`
`dir(ucollections)`

That is part of the main MicroPython port have you tried flashing your device using the upstream MicroPython port?  You can see instructions here -> [https://micropython.org/download/].  I hope that helps!