bbcmicrobit/micropython #685: Consistent failure when attempting to run the 'make all' command

AlasdairAtKitronik, 2020-10-05:
I was trying out the module freezing process (shown here: https://github.com/mytechnotalent/MicroPython-micro-bit_How_To_Build_And_Freeze_Modules), but kept having issues with the 'make all' process not working.

I thought it might be the changes I was making, so cloned a fresh version of the bbcmicrobit/micropython repo and tried to build the firmware with no changes - same error. I have tried using Windows command line and the git bash environment - both consistently fail on the same section (I am using a Windows 10 machine).

This is the error message:

```blocks
[96/325] Building CXX object ym/microbit-dal/source/CMakeF...s/microbit-dal.dir/drivers/MicroBitCompassCalibrator.cpp.o
FAILED: C:\PROGRA~2\GNUTOO~1\50A5A~1.420\bin\AR10B2~1.EXE -DYOTTA_MODULE_NAME=microbit-dal -Igenerated/include -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/microbit-dal -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/ble -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/ble-nrf51822 -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/microbit-dal/inc/core -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/microbit-dal/inc/types -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/microbit-dal/inc/drivers -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/microbit-dal/inc/bluetooth -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/microbit-dal/inc/platform -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/api -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/hal -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/cmsis -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/ble-nrf51822/source/btle -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/ble-nrf51822/source/btle/custom -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/ble-nrf51822/source/common -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/ble_radio_notification -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/ble_services/ble_dfu -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/common -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/device_manager -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/device_manager/config -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/peer_manager -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/device -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/ble_flash -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/delay -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/hal -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/pstorage -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/pstorage/config -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/bootloader_dfu -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/bootloader_dfu/hci_transport -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/crc16 -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/hci -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/scheduler -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/timer -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/util -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/fds -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/fstorage -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/experimental_section_vars -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/softdevice/common/softdevice_handler -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/softdevice/s130/headers -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/toolchain -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822 -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S110 -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822 -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/crc16 -IC:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/L [... cut ...]
C:/Users/alasdair.trench/github/microbit-micropython-no-mods/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:230:29: error: in C++98 'best' must be initialized by constructor, not by '{...}'
     Sample3D best = { 0,0,0 };
                             ^
[96/325] Building CXX object ym/microbit-dal/source/CMakeFiles/microbit-dal.dir/drivers/MicroBitAccelerometer.cpp.o
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
make: *** [yotta] Error 1
```
Is anyone able to shed light on why this fails? Are there extra steps which need to be taken before running the instructions in the README?
@dpgeorge @microbit-carlos

microbit-carlos, 2020-10-05:
What version of gcc are you using? Sounds like it's not compiling as C++11.

microbit-carlos, 2020-10-05:
Are you use the target was correctly set in Yotta?

AlasdairAtKitronik, 2020-10-05:
My gcc version is 8.1.0 and I used the following line to set the target:
```blocks
yt target bbc-microbit-classic-gcc-nosd
```

AlasdairAtKitronik, 2020-10-05:
I did think the references to C++98 were a bit strange

microbit-carlos, 2020-10-05:
Could you build with the verbose flags and copy the full output? that should show which flags were used.

microbit-carlos, 2020-10-05:
I just built on Ubuntu with gcc 7.3.1 without issues, these are the steps I take:

```
git clone https://github.com/bbcmicrobit/micropython.git
cd micropython
yt clean
yt target bbc-microbit-classic-gcc-nosd
yt up
make all
```

AlasdairAtKitronik, 2020-10-05:
This is the build with flags:
[build_flags.txt](https://github.com/bbcmicrobit/micropython/files/5326921/build_flags.txt)

AlasdairAtKitronik, 2020-10-05:
I followed exactly the same steps as you did for your Ubuntu build.

AlasdairAtKitronik, 2020-10-05:
This is my gcc version output:
```blocks
C:\Users\alasdair.trench\github\microbit-micropython-no-mods\micropython>gcc --version
gcc (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

AlasdairAtKitronik, 2020-10-05:
Is there an issue because I have the MinGW-W64, and it says at the start of the build that "This program built for i386-pc-mingw32"?

microbit-carlos, 2020-10-05:
I don't know sorry, maybe that is just the make program build info.

Based on the output though, sounds like you are building with gcc arm v5:

```
-- Found assembler: C:/Program Files (x86)/GNU Tools ARM Embedded/5.4 2016q3/bin/arm-none-eabi-gcc.exe
```

I'd recommend to update that.

This is your x86_64 gcc:
> C:\Users\alasdair.trench\github\microbit-micropython-no-mods\micropython>gcc --version
> gcc (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 8.1.0
> Copyright (C) 2018 Free Software Foundation, Inc. [...]

AlasdairAtKitronik, 2020-10-05:
Right, I'll try updating that - thanks

AlasdairAtKitronik, 2020-10-05:
The update of GCC solved my inital issue - thank you.
It might be worth noting in the README that as the make process uses a 'grep' command it will require a Unix environment?
All building successfully now.