bbcmicrobit/micropython #688: Publishing new firmware

aivarannamaa, 2020-10-14:
Congratulations on the new micro:bit!

Please also make the new hex available under releases! 

Thonny IDE currently depends on the JSON given by https://api.github.com/repos/bbcmicrobit/micropython/releases/latest to implement micro:bit firmware flasher. 

It would be nice if in the future the same URL gave recommended firmwares both for the old and new microbits. I would enhance the flasher so that it selects the proper firmware depending on the information in DETAILS.TXT.

microbit-carlos, 2020-11-27:
Hi Aivar

As this repo only contain source code the V1 port, we were not really planning to add the V2 hex files here.

The V2 port is current in https://github.com/microbit-foundation/micropython-microbit-v2/. The plan is to push the V2 port to the upstream MicroPython repository when it's ready, so this is a temporary repository until everything is upstream.

In the meantime we've pushed the first beta for V2 in https://github.com/microbit-foundation/micropython-microbit-v2/releases/tag/v2.0.0-beta.1. We are not planning to remove the repo in the future (maybe archive it?), so that URL should continue to work in the future, but when the port is upstream we might not be using the GH releases from that repo anymore.

aivarannamaa, 2020-11-27:
Thank you, Carlos!

Would it be OK, if I created a new dummy repository under https://github.com/thonny/ (eg. https://github.com/thonny/microbit) and manually mirrored the official releases there?

microbit-carlos, 2020-11-27:
Of course! If that's easier for Thonny to manage its hex downloads, then sounds like a good approach.
The only thing I'd suggest is that it might be good to make it clear in the repo that it's a mirror in case people encounter it randomly.

aivarannamaa, 2020-11-27:
> The only thing I'd suggest is that it might be good to make it clear in the repo that it's a mirror in case people encounter it randomly.

Sure!

closed 2020-11-27 by aivarannamaa

microbit-carlos, 2020-11-27:
@aivarannamaa btw, we also have a new format for hex files to be able to pack a V1 and V2 hex into a single file: https://tech.microbit.org/software/hex-format/

Maybe that'll be useful for Thonny? How does it transfer files to the micro:bit?

aivarannamaa, 2020-11-27:
@microbit-carlos, I was about to ask you about this.

I tried https://github.com/microbit-foundation/micropython-microbit-v2/releases/download/v2.0.0-beta.1/micropython-microbit-v2.0.0-beta.1.hex on my micro:bit v1.3 (the serial number starts with "9000"). The installation seemed to go well, but when I connected to the usual /dev/ttyACM0 with `tio`, I got no output (tried ENTER, Ctrl+C, Ctrl+B, Ctrl+D). 

When I unplugged and re-plugged the device, the volume got mounted, but /dev/ttyACM0 did not appear.

aivarannamaa, 2020-11-27:
> How does it transfer files to the micro:bit?

Thonny simply copies the hex to the device. It only supports "plain" MicroPython, ie. without embedded main script, so the user only installs a hex only when new version appears.

If the universal hex worked, it would be a good solution. But how would you version it? I understood that the MicroPython for the old micro:bit will not be updated.

microbit-carlos, 2020-11-27:
> I tried https://github.com/microbit-foundation/micropython-microbit-v2/releases/download/v2.0.0-beta.1/micropython-microbit-v2.0.0-beta.1.hex on my micro:bit v1.3 (the serial number starts with [...]

Right, that's because the v2 microbit-v2.0.0-beta.1.hex  files is only for micro:bit V2, as it's a different microcontroller (nRF52 with cortext M4 instead of the nRF51 with M0).

>When I unplugged and re-plugged the device, the volume got mounted, but /dev/ttyACM0 did not appear.

That shouldn't have really happened, the virtual serial com is provided by the interface chip, which shouldn't have been affected by flashing a hex to the nRF, even if interface chip cannot talk to to the nRF the tty should still be available. Is that completely gone, or are you able to see it after disconnecting and reconnecting the USB cable? 

> Thonny simply copies the hex to the device. It only supports "plain" MicroPython, ie. without embedded main script, so the user only installs a hex only when new version appears.

And how does it transfer the files into the micro:bit flash? Does it send the data via serial?

>If the universal hex worked, it would be a good solution. But how would you version it? I understood that the MicroPython for the old micro:bit will not be updated. Would it mean that the universal hex contains version 1.0.1 for old micro:bit and 2.0.0 for the new one?

We will continue to maintain this version of MicroPython, but not all the features from V2 will be able to make it to V1, as there is additional hardware and the underlaying abstraction layer has also changed. We will continue to work on V1, specially to see how to improve compatibility for scripts to work on both boards.

You can create a Universal Hex with this tool: https://tech.microbit.org/software/universal-hex-creator/

So in this case the hex file contains MicroPython v1.0.1 for micro:bit V1 and MicroPython v2.0.0-beta.1 for micro:bit V2.

I did the same today in https://github.com/mu-editor/mu/issues/1154#issuecomment-734859242, you can see the output file in https://github.com/mu-editor/mu/files/5608046/microbit-micropython_universal_v1.0.1_v2.0.0-beta.1.hex.zip.

aivarannamaa, 2020-11-27:
> Is that completely gone, or are you able to see it after disconnecting and reconnecting the USB cable?

Now I see it again. I don't know what it was, because I later had similar issue with another device as well.

> And how does it transfer the files into the micro:bit flash? Does it send the data via serial?

Yes, it writes series of commands like `_f = open(target, "w")` and `_f.write(b"....")` to the REPL.

> microbit-micropython_universal_v1.0.1_v2.0.0-beta.1.hex.zip.

If the version numbers remain different, then I guess I'll better go for separate HEX files and let Thonny use DETAILS.TXT to decide which one to download.

aivarannamaa, 2020-11-27:
> but /dev/ttyACM0 did not appear.

Sorry, I was just being stupid -- I had ssh-d into another computer and forgot about it :)

microbit-carlos, 2020-11-30:
> > And how does it transfer the files into the micro:bit flash? Does it send the data via serial?
> 
> Yes, it writes series of commands like `_f = open(target, "w")` and `_f.write(b"....")` to the REPL.

Right, in that case you could flash the universal hex independently of the micro:bit version and then transfer main.py (and any other python script) via the REPL.

However, I guess Thonny checks what version of MicroPython is on the board and if it's older than the version in Thonny it  flashes the board? If that's the case, then yes it might be easier to check the board ID (which can be done via pyserial or the details.txt) and then keep the individual Intel Hex files.

aivarannamaa, 2020-12-01:
> However, I guess Thonny checks what version of MicroPython is on the board and if it's older than the version in Thonny it flashes the board?

Actually it is simpler than that -- the user can select "Install firmware" and latest suitable MicroPython will be downloaded and installed regardless of what was on the board before.

microbit-carlos, 2020-12-02:
Right, so in that case if there is no version check the universal hex would work in any micro:bit board, no need to check the board ID.