bbcmicrobit/micropython #689: Building firmware

azamat-bagatov, 2020-10-30:
Hi everyone, I've been trying to build my own firmware to fix a sound-neopixel bug, fixed by this:
https://github.com/bbcmicrobit/micropython/pull/607/files

Now, this is a total joke, I've been trying to install yotta on 2 different Windows machines now, and got errors after errors, and fixes that created new errors and so on in a never-ending downward spiral.  It's been two days of trials and I'm nowhere near the end. 

Could anyone please build a firmware with that pull request for me? 
I would really appreciate it. 
Thanks.

microbit-carlos, 2020-11-02:
Hi @azamat-bagatov,

Could you post here the versions of the tools you've been using and the full output in the console? That would help us figure out the issues and make sure it works for everybody.

azamat-bagatov, 2020-11-02:
> Hi @azamat-bagatov,
> 
> Could you post here the versions of the tools you've been using and the full output in the console? That would help us figure out the issues and make sure it works for everybody.

Hi Carlos, thanks for getting back. So, I'm on Windows 10, Python 3.9. I did everything per [this guide](https://docs.yottabuild.org/#installing-on-windows) 
So far on this machine I'm stuck on "pip install -U yotta", when I run it it throws this error:
`C:\Users\User>pip install -U yotta
Requirement already up-to-date: yotta in c:\users\user\appdata\local\programs\python\python39\lib\site-packages (0.20.5)
Collecting semantic_version<3,>=2.3.1
  Using cached semantic_version-2.8.5-py2.py3-none-any.whl (15 kB)
Collecting requests[security]<3,>=2.4.3
  Using cached requests-2.24.0-py2.py3-none-any.whl (61 kB)
Collecting certifi>=2016.2.28
  Using cached certifi-2020.6.20-py2.py3-none-any.whl (156 kB)
Collecting PyGithub<2,>=1.25
  Using cached PyGithub-1.53-py3-none-any.whl (274 kB)
Collecting colorama<0.4,>=0.3
  Using cached colorama-0.3.9-py2.py3-none-any.whl (20 kB)
Collecting hgapi<2,>=1.7
  Using cached hgapi-1.7.4.tar.gz (13 kB)
Collecting Jinja2<3,>=2.7.0
  Using cached Jinja2-2.11.2-py2.py3-none-any.whl (125 kB)
Collecting cryptography<3,>=2.8
  Using cached cryptography-2.9.2.tar.gz (517 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting PyJWT<2.0,>=1.0
  Using cached PyJWT-1.7.1-py2.py3-none-any.whl (18 kB)
Collecting pathlib<1.1,>=1.0.1
  Using cached pathlib-1.0.1.tar.gz (49 kB)
Collecting jsonschema<3.0,>=2.4.0
  Using cached jsonschema-2.6.0-py2.py3-none-any.whl (39 kB)
Collecting argcomplete<2.0,>=0.8.0
  Using cached argcomplete-1.12.1-py2.py3-none-any.whl (38 kB)
Collecting mbed_test_wrapper<2.0.0,>=0.0.3
  Using cached mbed_test_wrapper-1.0.0.tar.gz (2.7 kB)
Collecting valinor<2.0,>=1.1.4
  Using cached valinor-1.1.4.tar.gz (12 kB)
Collecting intelhex<3.0,>=2.0
  Downloading intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
     |████████████████████████████████| 50 kB 539 kB/s
Collecting jsonpointer<2.0,>=1.9
  Using cached jsonpointer-1.14-py2.py3-none-any.whl (7.4 kB)
Collecting ntfsutils<0.2,>=0.1.3
  Using cached ntfsutils-0.1.4.tar.gz (7.3 kB)
Collecting chardet<4,>=3.0.2
  Using cached chardet-3.0.4-py2.py3-none-any.whl (133 kB)
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1
  Downloading urllib3-1.25.11-py2.py3-none-any.whl (127 kB)
     |████████████████████████████████| 127 kB 1.3 MB/s
Collecting idna<3,>=2.5
  Using cached idna-2.10-py2.py3-none-any.whl (58 kB)
Collecting pyOpenSSL>=0.14; extra == "security"
  Using cached pyOpenSSL-19.1.0-py2.py3-none-any.whl (53 kB)
Collecting deprecated
  Using cached Deprecated-1.2.10-py2.py3-none-any.whl (8.7 kB)
Collecting MarkupSafe>=0.23
  Using cached MarkupSafe-1.1.1.tar.gz (19 kB)
Collecting cffi!=1.11.3,>=1.8
  Using cached cffi-1.14.3-cp39-cp39-win_amd64.whl (179 kB)
Collecting six>=1.4.1
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Collecting pyyaml<6.0,>=5.1
  Using cached PyYAML-5.3.1.tar.gz (269 kB)
Requirement already satisfied, skipping upgrade: setuptools in c:\users\user\appdata\local\programs\python\python39\lib\site-packages (from valinor<2.0,>=1.1.4->yotta) (49.2.1)
Collecting pyOCD<1.0,>=0.3
  Downloading pyocd-0.28.2-py2.py3-none-any.whl (9.4 MB)
     |████████████████████████████████| 9.4 MB 3.3 MB/s
Collecting project_generator<0.9.0,>=0.8.0
[... 175 lines cut ...]
  warning: no previously-included files found matching '.travis'
  warning: no previously-included files matching '*' found under directory '.azure-pipelines'
  warning: no previously-included files matching '*' found under directory '.travis'
  warning: no previously-included files found matching 'release.py'
  warning: no previously-included files found matching '.coveragerc'
  warning: no previously-included files found matching 'codecov.yml'
  warning: no previously-included files found matching 'dev-requirements.txt'
  warning: no previously-included files found matching 'rtd-requirements.txt'
  warning: no previously-included files found matching 'tox.ini'
  writing manifest file 'src\cryptography.egg-info\SOURCES.txt'
  running build_ext
  generating cffi module 'build\\temp.win-amd64-3.9\\Release\\_padding.c'
  creating build\temp.win-amd64-3.9
  creating build\temp.win-amd64-3.9\Release
  generating cffi module 'build\\temp.win-amd64-3.9\\Release\\_constant_time.c'
  generating cffi module 'build\\temp.win-amd64-3.9\\Release\\_openssl.c'
  building '_openssl' extension
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 20.2.3; however, version 20.2.4 is available.
You should consider upgrading via the 'c:\users\user\appdata\local\programs\python\python39\python.exe -m pip install --upgrade pip' command.`
I tried a few things i found online, but non of them worked.

microbit-carlos, 2020-11-02:
Looks like some of the dependencies might not work in Python 3.9 yet. I'd recommend to use an older version of Python to install yotta at the moment (I'd try 3.6 or 3.7 in case 3.8 is not compatible either).

azamat-bagatov, 2020-11-02:
> Looks like some of the dependencies might not work in Python 3.9 yet. I'd recommend to use an older version of Python to install yotta at the moment (I'd try 3.6 or 3.7 in case 3.8 is not compatible either).

Thanks for this, it did work with python 3.6.9 . 
I'm following [this guide](https://microbit-micropython.readthedocs.io/en/v1.0.1/devguide/flashfirmware.html#flashfirmware)
However, when running everything the way it's described there, I get the following error:

```
PS C:\Users\User\Desktop\micropython-master> yt target bbc-microbit-classic-gcc-nosd
info: get versions for mbed-gcc
info: download mbed-gcc@0.1.3 from the public module registry
PS C:\Users\User\Desktop\micropython-master> yt up
info: get versions for bbc-microbit-classic-gcc-nosd
info: get versions for mbed-gcc
info: get versions for microbit-dal
info: update outdated: microbit-dal@2.1.1 -> microbit-dal@v2.1.1 from GitHub lancaster-university/microbit-dal
info: download microbit-dal@v2.1.1 from GitHub lancaster-university/microbit-dal
info: get versions for mbed-classic
warning: github rate limit for anonymous requests exceeded: you must log in

You need to log in with Github to do this.
[1mPress enter to continue.
[2mYour browser will open to complete login.[22m

waiting for response...
If you are unable to use a browser on this machine, please copy and paste this URL into a browser:
https://yottabuild.org/?provider=github#login/AAAAB3NzaC1yc2EAAAAEAAEAAQAAAQEAwb2erAQI3r%2B2qTLJXc8Mdg08U0Iot04LJ46mpr3mEY%2BmF22p4eHvbWdJPVsNFeTqq1tlqxnitEPtm5k7wGCQ5fEz7zOzcMGvTJHn6RJayq3o6m7O4ipZ6CPmkH8LG3OEg2Pheu%2BaPdmMSDf8DVBCwijJSIhTgM4/k58JdU2ykvN%2BvC2LxX7gk1iDfjEdkLPnK0VLyF10axXPsQ05pI/6M4ngYNbsBwTayh8NjjlEc80USaVXMYTAPJf3QaMOt7B41mLhKewJ9f3FJUSAKwrVw9kqyhHUEhXyCTKtiZyzHUs6tQ73Wr6jEuyZ1NKjEdVmnShzUnLN7gs5QNenJi6c6Q%3D%3D
.....
info: update outdated: mbed-classic@0.0.4 -> mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic
info: download mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic
info: get versions for ble
info: update outdated: ble@2.5.0 -> ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API
info: download ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API
info: get versions for ble-nrf51822
info: update outdated: ble-nrf51822@2.5.0 -> ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822
info: download ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822
info: get versions for nrf51-sdk
info: update outdated: nrf51-sdk@2.2.0 -> nrf51-sdk@v2.2.0+mb4 from GitHub lancaster-university/nrf51-sdk
info: download nrf51-sdk@v2.2.0+mb4 from GitHub lancaster-university/nrf51-sdk
PS C:\Users\User\Desktop\micropython-master> yt build
info: generate for target: bbc-microbit-classic-gcc-nosd 0.2.3 at C:\Users\User\Desktop\micropython-master\yotta_targets\bbc-microbit-classic-gcc-nosd
CMake Deprecation Warning at CMakeLists.txt:16 (cmake_policy):
  The OLD behavior for policy CMP0017 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:78 (cmake_force_c_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:79 (cmake_force_cxx_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

GCC version is: 4.9.3
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Tools ARM Embedded/4.9 2015q2/bin/arm-none-eabi-gcc.exe
[... 136 lines cut ...]
  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/User/Desktop/micropython-master/build/frdm-k64f-gcc/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/ninja/ninja.exe cmTC_ca95a && [1/2] Building C object CMakeFiles/cmTC_ca95a.dir/testCCompiler.c.o
    [2/2] Linking C executable cmTC_ca95a
    FAILED: cmd.exe /C "cd . && C:\PROGRA~2\GNUTOO~1\4636A~1.920\bin\AR19DD~1.EXE -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -Wl,-u,_printf_float -mcpu=cortex-m4 -mthumb -T"C:/Users/User/Desktop/micropython-master/yotta_targets/kinetis-k64-gcc/CMake/../ld/K64FN1M0xxx12.ld" -Wl,-Map,cmTC_ca95a.map -Wl,--start-group CMakeFiles/cmTC_ca95a.dir/testCCompiler.c.o    -lm -lc -lgcc -lm -lc -lgcc -Wl,--end-group  --specs=nano.specs -o cmTC_ca95a && cd ."
[... cut ...]
  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:149 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/User/Desktop/micropython-master/build/frdm-k64f-gcc/CMakeFiles/CMakeOutput.log".
See also "C:/Users/User/Desktop/micropython-master/build/frdm-k64f-gcc/CMakeFiles/CMakeError.log".
error: command ['cmake', '-D', 'CMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'Ninja', '.'] failed
PS C:\Users\User\Desktop\micropython-master>
```

microbit-carlos, 2020-11-02:
> The guide here describes this error and as a solution proposes installing gcc and specifying the target like this : yotta target frdm-k64f-gcc .

This will try to build the yotta project for a different board (the [Freedom-K64F](https://os.mbed.com/platforms/FRDM-K64F/)), so that will not work for the micro:bit.

From a very quick look at the output it looks like it's not building for C++11, are you sure you did a `yt clean` before the last `yt build` with the right microbit target?

The C++ standard version flag should be set up in the target, although maybe something is miss-configured?
Nevertheless in the meantime an easier workaround for you might be to use a newer GCC version. I've been mostly using v7.3.1:

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

azamat-bagatov, 2020-11-03:
> From a very quick look at the output it looks like it's not building for C++11, are you sure you did a `yt clean` before the last `yt build` with the right microbit target?

I did, both before and after "yt up". Unfortunately, trying this with both gcc 9.8.1 and 7.2.1 gives me the same old "command ninja failed" error. 
Here's a full output of my last attempt: 

```
PS C:\Users\User\Desktop\micropython-master> arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

PS C:\Users\User\Desktop\micropython-master> yt target bbc-microbit-classic-gcc-nosd
PS C:\Users\User\Desktop\micropython-master> yt up
info: get versions for bbc-microbit-classic-gcc-nosd
info: get versions for mbed-gcc
info: get versions for microbit-dal
info: update outdated: microbit-dal@2.1.1 -> microbit-dal@v2.1.1 from GitHub lancaster-university/microbit-dal
info: download microbit-dal@v2.1.1 from GitHub lancaster-university/microbit-dal
info: get versions for mbed-classic
info: update outdated: mbed-classic@0.0.4 -> mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic
info: download mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic
info: get versions for ble
info: update outdated: ble@2.5.0 -> ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API
info: download ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API
info: get versions for ble-nrf51822
info: update outdated: ble-nrf51822@2.5.0 -> ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822
info: download ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822
info: get versions for nrf51-sdk
info: update outdated: nrf51-sdk@2.2.0 -> nrf51-sdk@v2.2.0+mb4 from GitHub lancaster-university/nrf51-sdk
info: download nrf51-sdk@v2.2.0+mb4 from GitHub lancaster-university/nrf51-sdk
PS C:\Users\User\Desktop\micropython-master> yt clean
PS C:\Users\User\Desktop\micropython-master> yt build
info: generate for target: bbc-microbit-classic-gcc-nosd 0.2.3 at C:\Users\User\Desktop\micropython-master\yotta_targets\bbc-microbit-classic-gcc-nosd
CMake Deprecation Warning at CMakeLists.txt:16 (cmake_policy):
  The OLD behavior for policy CMP0017 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:78 (cmake_force_c_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:79 (cmake_force_cxx_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

GCC version is: 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc.exe
suppressing warnings from ble-nrf51822
suppressing warnings from nrf51-sdk
suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/User/Desktop/micropython-master/build/bbc-microbit-classic-gcc-nosd
[... 59 lines cut ...]
     if (p > RENDER_FRAMES) {
           ^
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c: In function 'Write':
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c:71:11: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (p > RENDER_FRAMES) {
           ^
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c: In function 'RenderSample':
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c:170:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
     if (X >= sizeof(tab48426))
     ^~
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c:172:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  mem53 = tab48426[X];
  ^~~~~
[129/325] Building C object source/CMakeFiles/microbit-mic.../User/Desktop/micropython-master/source/microbit/modos.c.o
[... cut ...] ource/microbit/modos.c
C:/Users/User/Desktop/micropython-master/source/microbit/modos.c:32:10: fatal error: genhdr/microbitversion.h: No such file or directory
 #include "genhdr/microbitversion.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[129/325] Building C object source/CMakeFiles/microbit-mic...C_/Users/User/Desktop/micropython-master/source/py/map.c.o
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
PS C:\Users\User\Desktop\micropython-master>

```

microbit-carlos, 2020-11-03:
Okay, so the newer GCC version has fixed the issues you were having before.
The problem from your last output might be from running "yt build" instead of "make all". Can you try that and see if it works?

azamat-bagatov, 2020-11-04:
> Okay, so the newer GCC version has fixed the issues you were having before.
> The problem from your last output might be from running "yt build" instead of "make all". Can you try that and see if it works?

Thank you. This is what I get. Isn't "make" a Unix thing anyway? 
```
PS C:\Users\User\Desktop\micropython-master> make all
make : The term 'make' is not recognized as the name of a cmdlet, function, script file, or operable program. Check
the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ make all
+ ~~~~
    + CategoryInfo          : ObjectNotFound: (make:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

microbit-carlos, 2020-11-10:
Make is a build tool that is also available on Windows (and something like [MinGW](http://www.mingw.org/) will also include other unix tools needed like grep) and is needed to build MicroPython.
If you'd prefer not to install it the Makefile is not too complex to figure out the commands needed for each step, but it would be a lot easier to use these tools.

azamat-bagatov, 2020-11-16:
> Make is a build tool that is also available on Windows (and something like [MinGW](http://www.mingw.org/) will also include other unix tools needed like grep) and is needed to build MicroPython.
> If [...]

Thanks Carlos, but I'm afraid make gives me the same thing... I'm starting to think the Universe for some reason doesn't want me to run neopixels with audio on microbit. Perhaps this is going to screw with the Matrix. Anyways, here's what the console gives me when running make all: 
```
PS C:\Users\User\Desktop\micropython-master> make all
python tools/makeversionhdr.py inc/genhdr/microbitversion.h
Generate inc/genhdr/microbitversion.h
info: generate for target: bbc-microbit-classic-gcc-nosd 0.2.3 at C:\Users\User\Desktop\micropython-master\yotta_targets\bbc-microbit-classic-gcc-nosd
CMake Deprecation Warning at CMakeLists.txt:16 (cmake_policy):
  The OLD behavior for policy CMP0017 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:78 (cmake_force_c_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:79 (cmake_force_cxx_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

GCC version is: 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc.exe
suppressing warnings from ble-nrf51822
suppressing warnings from nrf51-sdk
suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/User/Desktop/micropython-master/build/bbc-microbit-classic-gcc-nosd
[61/325] Building CXX object ym/microbit-dal/source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitIOPinService.cpp.o
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp: In member function 'void MicroBitIOPinService::onDataWritten(const GattWriteCallbackParams*)':
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:179:42: warning: array subscript is above array bounds [-Warray-bounds]
                 io.pin[i].getDigitalValue();
                 ~~~~~~~~~~~~~~~~~~~~~~~~~^~
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:182:41: warning: array subscript is above array bounds [-Warray-bounds]
                 io.pin[i].getAnalogValue();
                 ~~~~~~~~~~~~~~~~~~~~~~~~^~
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:199:41: warning: array subscript is above array bounds [-Warray-bounds]
                io.pin[i].getDigitalValue();
                ~~~~~~~~~~~~~~~~~~~~~~~~~^~
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:202:40: warning: array subscript is above array bounds [-Warray-bounds]
                io.pin[i].getAnalogValue();
                ~~~~~~~~~~~~~~~~~~~~~~~~^~
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:224:43: warning: array subscript is above array bounds [-Warray-bounds]
                 io.pin[pin].setAnalogValue(value);
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:225:46: warning: array subscript is above array bounds [-Warray-bounds]
                 io.pin[pin].setAnalogPeriodUs(period);
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:245:51: warning: array subscript is above array bounds [-Warray-bounds]
                  io.pin[data->pin].setDigitalValue(data->value);
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:247:50: warning: array subscript is above array bounds [-Warray-bounds]
                  io.pin[data->pin].setAnalogValue(data->value == 255 ? 1023 : data->value << 2);
[... 98 lines cut ...]
  mem53 = tab48426[X];
  ^~~~~
[... cut ...] ngth=44 && cd /D C:/Users/User/Desktop/micropython-master/build/bbc-microbit-classic-gcc-nosd/source && "C:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-objcopy.exe" -O binary microbit-micropython microbit-micropython.bin""
'srec_cat' is not recognized as an internal or external command,
operable program or batch file.
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
make: *** [yotta] Error 1
PS C:\Users\User\Desktop\micropython-master>
```

microbit-carlos, 2020-11-16:
Almost there!
The dal also needs srecord, more info in https://lancaster-university.github.io/microbit-docs/offline-toolchains/#installation-on-windows.

azamat-bagatov, 2020-11-19:
> Almost there!
> The dal also needs srecord, more info in https://lancaster-university.github.io/microbit-docs/offline-toolchains/#installation-on-windows.

All right, I think I'm getting there. It did build everything, but when I copy the hex file in micropython-master\build\bbc-microbit-classic-gcc-nosd\source to MAINTENANCE drive, I get txt file FAIL.TXT with:
`In application programming write failed.`
Here's the content of the details.txt file:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0000000051864e4500631011000000220000000097969901
HIC ID: 97969901
Auto Reset: 0
Automation allowed: 1
Overflow detection: 0
Daplink Mode: Bootloader
Bootloader Version: 0243
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD
Bootloader CRC: 0x32eb3cfd
Interface CRC: 0x07911068
Remount count: 1
```
And just in case here's the console output for "make all": 

```
PS C:\Users\User\Desktop\micropython-master> make all
python tools/makeversionhdr.py inc/genhdr/microbitversion.h
Generate inc/genhdr/microbitversion.h
info: generate for target: bbc-microbit-classic-gcc-nosd 0.2.3 at C:\Users\User\Desktop\micropython-master\yotta_targets\bbc-microbit-classic-gcc-nosd
CMake Deprecation Warning at CMakeLists.txt:16 (cmake_policy):
  The OLD behavior for policy CMP0017 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:78 (cmake_force_c_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

CMake Deprecation Warning at C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  C:/Users/User/Desktop/micropython-master/yotta_targets/mbed-gcc/CMake/toolchain.cmake:79 (cmake_force_cxx_compiler)
  toolchain.cmake:8 (include)
  C:/Program Files/CMake/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:75 (project)

GCC version is: 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc.exe
suppressing warnings from ble-nrf51822
suppressing warnings from nrf51-sdk
suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/User/Desktop/micropython-master/build/bbc-microbit-classic-gcc-nosd
[31/325] Building CXX object ym/microbit-dal/source/CMakeFiles/microbit-dal.dir/bluetooth/MicroBitIOPinService.cpp.o
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp: In member function 'void MicroBitIOPinService::onDataWritten(const GattWriteCallbackParams*)':
C:/Users/User/Desktop/micropython-master/yotta_modules/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp:179:42: warning: array subscript is above array bounds [-Warray-bounds]
                 io.pin[i].getDigitalValue();
[... 115 lines cut ...]
C:/Users/User/Desktop/micropython-master/source/py/emitinlinethumb.c: In function 'emit_inline_thumb_end_pass':
C:/Users/User/Desktop/micropython-master/source/py/emitinlinethumb.c:97:75: warning: unused parameter 'type_sig' [-Wunused-parameter]
 STATIC void emit_inline_thumb_end_pass(emit_inline_asm_t *emit, mp_uint_t type_sig) {
                                                                           ^~~~~~~~
[324/325] Building C object source/CMakeFiles/microbit-mic.../User/Desktop/micropython-master/source/lib/sam/render.c.o
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c: In function 'Read':
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c:51:11: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (p > RENDER_FRAMES) {
           ^
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c: In function 'Write':
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c:71:11: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (p > RENDER_FRAMES) {
           ^
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c: In function 'RenderSample':
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c:170:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
     if (X >= sizeof(tab48426))
     ^~
C:/Users/User/Desktop/micropython-master/source/lib/sam/render.c:172:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  mem53 = tab48426[X];
  ^~~~~
[325/325] Linking CXX executable source/microbit-micropython
'objdump' is not recognized as an internal or external command,
operable program or batch file.
make: *** [build/veraddr.txt] Error 255
```

microbit-carlos, 2020-11-19:
The MAINTENANCE drive is accessed when pressing the reset button while plugging the USB cable, this is a special mode to update the USB microcontroller, not the main microcontroller in the micro:bit.

You just need to copy the hex file into the MICROBIT drive.

azamat-bagatov, 2020-11-22:
And IT WORKS! The bug is gone! Thank you so much @microbit-carlos , you're a legend, I really appreciate your help!

closed 2020-11-22 by azamat-bagatov