bbcmicrobit/micropython #690: Supporting RGB+W in the Neopixel library.

ssav7912, 2020-11-06:
Hi, first query on Github so I apologise for any missteps I make. 

I've been looking to use one of adafruits 16 pixel WS2812 RGB+W neopixel rings in a project, and find that it behaves strangely when using it on the microbit. I realise that this is because this micropython microbit neopixel library only supports RGB, so the data sent to the ring is undershooting by a byte(?). I would like to make changes to the library in order to support RGB+W - I don't mind terribly if there's no conditional handling for 3 or 4 channels and such.

The assembler function in the source ([neopixelsend.s](https://github.com/bbcmicrobit/micropython/blob/master/source/lib/neopixelsend.s)), has comments that make references to needed changes to support RGB+W, but I'm not sure what they are, having only an abstract grasp of assembler. My suspicion is that I should be able to add an extra channel in the [modneopixel.cpp](https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/modneopixel.cpp) file and it should just send the appropriate data to r5 (num. LEDs * 4 rather than 3), to count to zero from, or does the assembly assume 3 channels per pixel throughout and doing this will break things?

ssav7912, 2020-11-08:
I have dug around further and answered my own question. The assembly does not assume 3 channels throughout and I have written a patch to allow 4 channel WS2812 neopixels to be used. See: #691

dpgeorge, 2021-11-29:
Now implemented by 652116dd094d241807b0e16b8b51847c5eb469fa

closed 2021-11-29 by dpgeorge