bbcmicrobit/micropython #693: pin5.write_digital(0) ValueError: Pin 5 in button mode

finalvalue, 2020-12-03:
pin5.write_digital(0)

ValueError: Pin 5 in button mode

microbit-carlos, 2020-12-23:
Hi @finalvalue,

Pin 5 is connected to button A on the micro:bit, which is why it's configured in button mode. If you other pins available is recommendable to use those before the pins shared with other micro:bit functionality (like the buttons or the display).

You can find more info about the pins in the documentation: https://microbit-micropython.readthedocs.io/en/latest/pin.html

finalvalue, 2020-12-24:
Yes, the current problem is that the hardware uses P5 and P11 pins, which can be used in makecode but not in micropython. 

Change the hardware later,。

@microbit-carlos  thanks for the reply！

closed 2020-12-24 by finalvalue