bbcmicrobit/micropython #699: [v2] i2s.scan got freezed

xmeow, 2020-12-24:
The whole firmware is stuck if I try to run `i2c.scan()` with some peripheral connected to it. Nothing I can do except resetting the board. But read and write actually works.
Thanks for the great upgrade of microbit, here is some terminal output hope it will help.
```
>>> MicroPython v1.13 on 2020-11-24; micro:bit v2.0.0-beta.1 with nRF52833
Type "help()" for more information.
>>> 
>>> i2c.
__class__       read            write           init
scan
>>> i2c.write(0x40, bytearray([0,0]))
>>> i2c.scan() # freeze here
```

carlosperate, 2020-12-30:
Thanks for reporting this issue.
As the V2 MicroPython is a different port development is currently carried out in https://github.com/microbit-foundation/micropython-microbit-v2.

This should have been resolved as part of https://github.com/lancaster-university/codal-microbit-v2/issues/49 and be fixed in the latest beta from https://github.com/microbit-foundation/micropython-microbit-v2/releases.

Could you give that a test to see if it works now?

xmeow, 2021-01-04:
@carlosperate Thanks for advise the correct repo for micropython firmware.  The only issue is that the latest beta.3 firmware may output unambiguous devices that don't exist.

```
MicroPython v1.13 on 2020-12-21; micro:bit v2.0.0-beta.3 with nRF52833
Type "help()" for more information.
>>> i2c.scan()
[9, 12, 15, 17, 19, 21, 23, 26, 29, 31, 33, 35, 37, 39, 42, 44, 47, 49, 51, 53, 55, 58, 60, 63, 65, 66, 68, 71, 73, 75, 77, 79, 81, 83, 86, 88, 91, 93, 96, 98, 100, 102, 104, 107, 109, 111, 113, 114, 117, 119]
>>>
```
I tried to build the firmware locally and change the last stop flag to false and got the correct output. 
```
>>> MicroPython v1.13-dirty on 2021-01-04; micro:bit v2.0.0-beta.3 with nRF52833
Type "help()" for more information.
>>>
>>> i2c.scan()
[64, 112]

```
It looks like the last param should be named repeat instead of stop.
```
STATIC mp_obj_t microbit_i2c_scan(mp_obj_t self_in) {
    mp_obj_t list = mp_obj_new_list(0, NULL);
    // 7-bit addresses 0b0000xxx and 0b1111xxx are reserved
    for (int addr = 0x08; addr < 0x78; ++addr) {
        int ret = microbit_hal_i2c_writeto(addr, NULL, 0, false); // true to flase
```

closed 2021-01-04 by xmeow

microbit-carlos, 2021-01-04:
Thank you @xmeow, I've opened issue https://github.com/microbit-foundation/micropython-microbit-v2/issues/55 to look into this.