bbcmicrobit/micropython #700: ValueError: invalid period

hnwangkg-ezio, 2021-01-08:
打扰了，我有个问题，我的代码是这样写的，：
from microbit import *
pin0.set_analog_period(20)

然后下载进micro bit V2
打印的logo是这样的：
croPython v1.13 on 2020-12-21; micro:bit v2.0.0-beta.3 with nRF52833
Type "help()" for more information.
>>> import microbit
>>> microbit.reset()
Traceback (most recent call last):
  File "main.py", line 86, in <module>
  File "main.py", line 18, in __init__
VaTraceback (most recent call last):
  File "main.py", line 86, in <module>
  File "main.py", line 18, in __init__
ValueError: invalid period

但是micro bit V1的正常

hnwangkg-ezio, 2021-01-08:
Excuse me, I have a question. My code is written like this:
from microbit import *
pin0.set_analog_period(20)

Then download into micro bit V2
The printed logo looks like this:
croPython v1.13 on 2020-12-21; micro:bit v2.0.0-beta.3 with nRF52833
Type "help()" for more information.
>>> import microbit
>>> microbit.reset()
Traceback (most recent call last):
   File "main.py", line 86, in <module>
   File "main.py", line 18, in __init__
VaTraceback (most recent call last):
   File "main.py", line 86, in <module>
   File "main.py", line 18, in __init__
ValueError: invalid period

But the micro bit V1 is normal

microbit-carlos, 2021-02-01:
Hi @hnwangkg-ezio,

It looks like you script had more code and there was an invalid period (`.`) in line 18. Do you have the full code?

microbit-carlos, 2022-08-08:
Closing due to lack of activity, hopefully this was just an issue in the coding having invalid syntax.
@hnwangkg-ezio if you are still experiencing this feel free to reopen it and we can look into this.

closed 2022-08-08 by microbit-carlos