bbcmicrobit/micropython #702: Add bluetooth for microbitsV2

pakal, 2021-01-09:
The docs mention that micropython and BLE stack can't be loaded together, but the microbits V2 seems to have far enough memory for that. Would bluetooth support be possible in micropython now?

https://microbit-micropython.readthedocs.io/en/latest/ble.html

microbit-carlos, 2021-02-01:
Hi @pakal,

Thanks for the question. We are currently updating the Bluetooth page in https://github.com/bbcmicrobit/micropython/pull/705. It'd be great if you could have a read as well to make sure things are clear enough.

A similar question has also been asked in the MicroPython for micro:bit V2 repository: microbit-foundation/micropython-microbit-v2#51
We can continue the conversation over there :)

closed 2021-02-01 by microbit-carlos

JLFra, 2021-03-18:
Hello,

For you information, Standard Micropython provide BLE. it works perfectly on ESP32 and STM32 WB55.
http://icnisnlycee.free.fr/index.php/61-stm32/ble-en-micropython
So do when will it be possible om Micropython Microbit V2?