bbcmicrobit/micropython #703: Can't install BBC micro:bit MicroPython

yufeiluo, 2021-01-25:
OS: debian10.
[code]
    uname -a
    Linux mydebian 4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux
[/code]
Do as the manual webpage say to install microbit module.

https://microbit-micropython.readthedocs.io/en/latest/devguide/installation.html
[code]
    sudo add-apt-repository -y ppa:team-gcc-arm-embedded
    sudo add-apt-repository -y ppa:pmiller-opensource/ppa
    sudo apt-get update
    sudo apt-get install cmake ninja-build gcc-arm-none-eabi srecord libssl-dev
    pip3 install yotta
[/code]
Building firmware:
[code]
    yt target bbc-microbit-classic-gcc-nosd
[/code]
Run yotta update to fetch remote assets:
[code]
    yt up
[/code]
Now it encounter an error info:

    **error: No module.json file.
    error: The current directory does not contain a valid module.**

How can i fix it?

microbit-carlos, 2021-02-01:
Hi @yufeiluo,

There is no need to create a `module.json` to build MicroPython, you can just follow the instructions from the [README.md](https://github.com/bbcmicrobit/micropython/blob/master/README.md).

microbit-carlos, 2022-03-18:
Hi @yufeiluo, were you able to resolve this issue?

microbit-carlos, 2022-07-04:
Closing due to lack of activity. If anybody is still experiencing this issue feel free to reopen it and we can look into it 👍

closed 2022-07-04 by microbit-carlos