bbcmicrobit/micropython #714: PCA9685 Error

lzty634158, 2021-03-23:
def __init__(self, i2c, address=PCA9685_ADDRESS):
        """Initialize the PCA9685."""
        self.address = address
        i2c.write(self.address, bytearray([MODE1, RESET]))
        # reset not sure if needed but other libraries do it
        self.set_all_pwm(0, 0)
        i2c.write(self.address, bytearray([MODE2, OUTDRV]))
        i2c.write(self.address, bytearray([MODE1, ALLCALL]))
        sleep(5)  # wait for oscillator
        i2c.write(self.address, bytearray([MODE1]))
        # write register we want to read from first
        mode1 = i2c.read(self.address, 1)
        mode1 = ustruct.unpack('<H', mode1)[0]
        mode1 = mode1 & ~SLEEP  # wake up (reset sleep)
        i2c.write(self.address, bytearray([MODE1, mode1]))
        sleep(5)  # wait for oscillator

Error:  mode1 = ustruct.unpack('<H', mode1)[0]
ValueError: buffer too small

microbit-carlos, 2021-04-20:
Hi @lzty634158,

It's possible `i2c.read(self.address, 1)` is returning an empty byte array, and `ustruct` cannot unpack it. You can try printing `mode1` to see what it contains before the unpacking.

microbit-carlos, 2022-08-08:
Closing due to lack of activity. @lzty634158 if you are still experiencing this issue please feel free to reopen this issue and we can have a look at it 👍

closed 2022-08-08 by microbit-carlos