bbcmicrobit/micropython #721: microbit hang out when using radio after some minutes

carlosmoca, 2021-07-20:
Dear all,

I'm envolved in a project that uses micro:bit to send data from a point to another point, where there is a second microbit, by using radio comm.

Since the required distance is about 50 m, the radio power is set in 7. The code performances  sensors readings and send the data every 60 s.

The issue is that in a period between 20 and 120 minutes (more and less), the emiter gets blocked. I have tested using differents power sources, solar-batteries, batery of 6 and 12 V (by using a 3,3 regulator and powering by the 3 V connector), and by a power source at 5 V by the mini-USB connector.

I've placed 2 capacitor of 100 uF (electrolitic) and 100 nF (ceramic) to supply the power when the radio occurs. I tried to see with the oscilloscope how big is the peak that occurs when the radio communication, but I could not see any relevant consume. 

What I've seen is that when voltage reduces lower than 2.4 (aprox), the mbit enters in this "sleep" o blocked mode and even if the voltage goes up again, it is not able to recover by itself without reseting it.

At the beginning, when I was using batteries, I though the mbit was consuming all the charge quite fast because of these peaks, but that was discarded when I checked voltages and they were ok. Actually, when the mbit stops working and you go and check the input voltage, it is OK( either 3,3 o 5, depending of the input I'm using in that moment), so the issue has to occur in a very short moment, but enough to put the mbit in that state.

I was suspecting that there was a peak of current that did the power source reduced the voltage because of it internal impedance, and so the voltage on the mbit were down lower than 2.5. But it is strange that this happen with such different power sources.

I'm using mbit V2, with the last firmaware updated on them.

ukBaz, 2021-07-21:
I wonder if this question would be better on the microbit community Slack:

https://support.microbit.org/support/solutions/articles/19000016193-microbit-community-slack

microbit-mark, 2021-07-21:
Thanks for raising this @carlosmoca. Do you have a short reproducer program for the issue? It would be really handy to have:

- Short reproducer program (.py script and .hex file)
- Details on how you have measured that the issue occurs at 2.4V
- Details of your hardware setup eg photo of the connections to the power source
-  Whether you see any change in the power LED when the “sleep” issue occurs.
- DETAILS.TXT

It would be useful to see if the display is refreshing when the issue occurs, so we know if the program is still running or if the firmware is blocked. Could you try the same program including something on the display and let me know if the LEDs remain on when the issue occurs?

carlosmoca, 2021-07-21:
> Thanks for raising this @carlosmoca. Do you have a short reproducer program for the issue? It would be really handy to have:
>
> * Short reproducer program (.py script and .hex file)
> * Details on how [...]

Dear Mark,

Thanks for your answer.

Right now I'm using 2 Micro:bits V2, one of them is the emiter, wich send several data by radio to the second, which has an IOT:bit, and that send the data to the cloud.

The problem is in the emiter. I added a message to check if the microbit was blocked or just the radio, and I could check that when the issue occurs, the display doesn't show data, so the issue affects to the micro bit and not just the radio.

The emiter has connected several sensors and also an OpenLog to store data. It collects data every 60 seconds and sends to the receiver.

I've tested several power systems, by 3,3 V input with batteries, solar panels and power source, and by 5 V microUSB input with same systems. In all the cases the failure occurs. When I check the voltage I measure the voltage at the input of the microbit, checking that when it is in this "sleep" mode the input voltage is right.

I'm testing now a simple code that just send a value every 60 second, and at the moment, it's working for several hours without problems. So my idea is to add new code step to step to the final code to identify where the error is, and if it is a programming error rather a hardware error. I'm concerned about how I send several data by radio, by using several radio send value... and if this could be causing the issue.

I attach the .hex of the emiter. As you can see, there is some code to read the sensors, store the data on the open log, and send the data by radio. The function Send_to_Router contains the code to send by radio the several parametters collected by the sensors in the other part of the code.

carlosmoca, 2021-07-21:
Dear Mark,

Thanks for your answer.

Right now I'm using 2 Micro:bits V2, one of them is the emiter, wich send several data by radio to the second, which has an IOT:bit, and that send the data to the cloud.

The problem is in the emiter. I added a message to check if the microbit was blocked or just the radio, and I could check that when the issue occurs, the display doesn't show data, so the issue affects to the micro bit and not just the radio.

The emiter has connected several sensors and also an OpenLog to store data. It collects data every 60 seconds and sends to the receiver.

I've tested several power systems, by 3,3 V input with batteries, solar panels and power source, and by 5 V microUSB input with same systems. In all the cases the failure occurs. When I check the voltage I measure the voltage at the input of the microbit, checking that when it is in this "sleep" mode the input voltage is right.

I'm testing now a simple code that just send a value every 60 second, and at the moment, it's working for several hours without problems. So my idea is to add new code step to step to the final code to identify where the error is, and if it is a programming error rather a hardware error. I'm concerned about how I send several data by radio, by using several radio send value... and if this could be causing the issue.

I attach the .hex of the emiter. As you can see, there is some code to read the sensors, store the data on the open log, and send the data by radio. The function Send_to_Router contains the code to send by radio the several parametters collected by the sensors in the other part of the code.

[microbit-Emisor-10_07_2021 (15).txt](https://github.com/bbcmicrobit/micropython/files/6857664/microbit-Emisor-10_07_2021.15.txt)
[Python_Code.txt](https://github.com/bbcmicrobit/micropython/files/6857673/Python_Code.txt)

microbit-mark, 2021-07-22:
Thanks for the follow-up. The code you have send across is using MakeCode Python rather than MicroPython. As such this issue should be raised on https://github.com/Microsoft/pxt-microbit/issues

You can read more about the difference between MakeCode Python and MicroPython in this article https://support.microbit.org/en/support/solutions/articles/19000111744-makecode-python-and-micropython

> I'm testing now a simple code that just send a value every 60 second, and at the moment, it's working for several hours without problems. So my idea is to add new code step to step to the final code [...]

This sounds like an appropriate way to debug the issue. If you keep adding a code step until it errors and then raise the issue on https://github.com/Microsoft/pxt-microbit/issues or via emailing help@microbit.org, we'll be able to help more there.

microbit-carlos, 2022-08-08:
I'll close this, as it is an issue in MakeCode Python and not MicroPython. As Mark mentioned, MakeCode issues can be reported in https://github.com/Microsoft/pxt-microbit/issues, thanks!

closed 2022-08-08 by microbit-carlos