bbcmicrobit/micropython #722: tutorials→music: Speaker, really?

msalat, 2021-07-30:
https://github.com/bbcmicrobit/micropython/blob/v1.0.1/docs/tutorials/music.rst

> It's very easy to generate bleeps and bloops from the device if you **attach a speaker**.

Tried it, micro:bit survived it. Then read the micro:bit schematic & NRF52833 datasheet: 
 * https://infocenter.nordicsemi.com/pdf/nRF52833_PS_v1.3.pdf#page=145
 * RING0 is directly attached to P0.02/AIN0, "standard drive".  
 * "standard drive" = 4 mA sink/source. Typical Speakers have 4…8 Ohms. (3.3 V (=VDD, worst case) -0.4V) / 4 Ohms (worst case) = 0.725 A. Factor 181.
 * Did not find anything about "short circuit protection" or similar.
 * Is attaching a speaker directly a good idea?

msalat, 2021-07-30:
https://infocenter.nordicsemi.com/pdf/nRF52832_PS_v1.3.pdf#page=156 (nRF5283**2**) max **sink** current is 7 mA @ VDD=3.3V.

rhubarbdog, 2022-07-13:
Extrenal speakers and microbit is a bit suckling em and see.
A real magnet +coil speaker requiers a reverse bias diode