bbcmicrobit/micropython #749: Blocking interupts and using the display```

rhubarbdog, 2022-05-22:
I have a DHT11 thermometer, it only works with microbit micropython is as an assembly routine with blocked interupts.
if i use a `print` statement to show the result it works ok, if i use `display.scroll` there is a 15 second delay between the program starting and reading the sensor and the result being shown on the microbit display.

```from microbit import *

@micropython.asm_thumb
def block_irq():
    cpsid('i')          # disable interrupts to go really fast                  

@micropython.asm_thumb
def unblock_irq():
    cpsie('i')          # enable interupts nolonger time critical               

block_irq()
# read sensor                                                                   
sleep(1000)
unblock_irq()
print('Microbit')```

this program displays the word `Microbit` after 1 second when ran from a serial terminal, where as the following program takes about 15 seconds to display the word `Microbit` on the led display.

```from microbit import *

@micropython.asm_thumb
def block_irq():
    cpsid('i')          # disable interrupts to go really fast                  

@micropython.asm_thumb
def unblock_irq():
    cpsie('i')          # enable interupts nolonger time critical               

block_irq()
# read sensor                                                                   
sleep(1000)
unblock_irq()
display.scroll('Microbit')```

is there any way to use the display if interrupt blocking is required?

microbit-carlos, 2022-08-08:
@dpgeorge this is not a blocker for the v1.1 release, but if it's an easy fix it'd be nice for get it in for the next release.

dpgeorge, 2022-08-21:
I can't reproduce this issue.  Trying with a V1 micro:bit (does it matter which hardware revision??) and the build from actions, microbitv1-micropython-2022-08-19-8d2138720fa2cdadbeebba8a8f0a72563b891f20.hex.zip, the above code by @microbit-carlos works OK for me.  It works with a sleep of 10, 20 and 1000.  It does not get stuck before the `display.scroll()`.

microbit-carlos, 2022-08-23:
I can confirm that this is no longer a problem in the latest MicroPython CI builds.
Just to double check I tried it again in the Python Editor v1.0.1 release and replicates there, so we can close this as fixed in the next release 👍

closed 2022-08-23 by microbit-carlos