bbcmicrobit/micropython #750: code example asmleds.py doesn't work on version 2 microbits

rhubarbdog, 2022-05-22:


microbit-carlos, 2022-06-24:
Hi @rhubarbdog,

As this example is using assembly code it is tuned specifically to the V1 microcontroller (nRF51822 Cortex-M0+) and pinout. The V2 microcontroller is different (nRF52833 Cortex-M4f) and the LEDs are connected to different pins, so the example might need tweaking to work for V2.

I think in this specific case, the only thing that might need changing is the pins, and to take in consideration that the V1 display is a 6x3 matrix, and in V2 it is a 5x5.

If you'd like to update the script to make it compatible with both boards and submit a PR we'd be happy to review it and merge it!

rhubarbdog, 2022-06-24:
I used that code to get a full dht11 thermometer going,  where do I find
the addresses

Phil

microbit-carlos, 2022-06-24:
You can get info about the pinout here: https://tech.microbit.org/hardware/edgeconnector/
Or looking at the schematics here:
- https://github.com/bbcmicrobit/hardware
- https://github.com/microbit-foundation/microbit-v2-hardware

microbit-carlos, 2022-08-08:
Hi @rhubarbdog, were you able to resolve this? Can this issue be closed?

rhubarbdog, 2022-08-09:
Please leAve this open. I still need the memory address for the gpio pins and to actually do the coding

rhubarbdog, 2022-11-07:
Hi without further input from say @dpgeorge on the new memory addresses for the rows I'm unable to take this any further.
Do we need a asmleds.py for version 2 microbits

dpgeorge, 2022-11-07:
The v2 micro:bit has two GPIO ports (compared to one on v1).  Their base address is:
- GPIO0: 0x50000500 (same as GPIO0 on v1)
- GPIO1: 0x50000800

Both these ports have the same register layout as v1.  So the code to toggle LEDs in assembler is the same for v1 and v2.

The main difference with v2 is that the columns and rows are on completely different pins to v1, and they are not continuous.  They pins are (eg P0.28 means GPIO0, pin 28):
- columns: P0.28, P0.11, P0.31, P1.05, P0.30
- rows: P0.21, P0.22, P0.15, P0.24, P0.19

On v1 these were all continuous so it was easy to cycle through the columns/rows just by shifting bits.  But on v2 it'll be more complicated, it won't be a simple loop.

dpgeorge, 2022-11-10:
I tested that code and indeed it does not work as expected.

I'm not sure exactly what is wrong, but it may be that you need to write to DIRSET (0x50000518) to set the pins as output (set the bit 1 to make that pin output).

Note that you can also use `machine.mem32[0x50000508] |= 1 << pin` to toggle a pin on.  It may be easier to prototype things using `mem32` instead of inline assembler.

rhubarbdog, 2022-11-10:
i'm not going to put any more effort in for now, if i get near a solution in another project i'll give it a whirl

closed 2022-11-10 by rhubarbdog