bbcmicrobit/micropython #752: one wire and micro:bit V2 for ds18b20 sensor ?

G-Boronad, 2022-06-01:
Hy, 
First, thanks for all your work!

I'm looking for a solution  to make works  ds18b20 sensor with,  one wire protocol and the micro:bit V2 (and micropyton of course).  
Is it possible? 
Thanks
G.B

rhubarbdog, 2022-07-08:
Hi,
You'll need to use arm 2 assembly that's the best I can give.
Check my dht11 repository to help but that's for v1 microbiology.  Yet to update to v2

Phil

rhubarbdog, 2022-10-01:
hi i've finished my dht11 repo and have decided to try and get this sensor working. i have just bought a ds18b20 so watch this space ...

G-Boronad, 2022-10-02:
Thanks a lot !!!!!!!

microbit-carlos, 2022-10-19:
I'll close this as a duplicate of https://github.com/microbit-foundation/micropython-microbit-v2/issues/99.

Thanks @rhubarbdog for working on porting the driver to micro:bit V2! If you want to share the driver in https://github.com/microbit-foundation/micropython-microbit-v2/issues/99 when it's done that would be fantastic 😃

closed 2022-10-19 by microbit-carlos