bbcmicrobit/micropython #755: cannot get predicatble communicaiton with radio

segrovets, 2022-06-22:
Hello, 

I'm writing a script that involves communication between micro:bit devices. I want to at a later time isolate which m:b's on the network receive certain communications. So I was experimenting with manually setting the channel configuration. (if there is a better way to do this, e.g via setting address please let me know)

My issue is that my micro:bits cannot communicate to each other if set to the same channel. To replicate:

I have 2 m:b V2 devices and one m:b V1.
All devices are connected to my pc via micro-usb cable.

I am using vs-code with the micro:bit python extension by MAKintact, seems to work great. 

The issue (and my attempts at debugging):
set channel = 0
m:b V2 devices can NOT communicate to each other
m:b V1 device will receive messages from both V2's 

When I set channel = 1 only on m:b V2-B
m:b V2-B can send/receive to/from m:b V2-A
m:b V2-B can receive from m:b V1, but not send to V1
m:b V2-A can send to both V1 and m:b V2-B, but cannot receive from V1

When I set both V2's one channel higher than the V1,
V1 can send to both V2's, but nobody can receive signal from V2 A or B

When they are one channel apart V1:ch6 V2Ach7 V2Bch8
A and b can send/receive
V1 can send to A but receives from no one

If i set V1 to ch7 and V2Ach6 and V2Bch8
V1 can send to both and no one else can communicate.

My only conclusion is that there is a miss-match in how these different micro:bits are interpreting channel information!
What to do! please help.

[minimal_working_error.txt](https://github.com/bbcmicrobit/micropython/files/8955057/minimal_working_error.txt)

rhubarbdog, 2022-07-12:
Hi is it all micro python or blocks and micro python?

segrovets, 2022-08-08:
@microbit-carlos 
This code does work as expected, when used with python.microbit.org/v/beta

I am still not sure why my other code doesn't work. But I will use this online IDE as a starting point.

Thank you.
segrovets

microbit-carlos, 2022-08-08:
Hi @segrovets, thanks for confirming the test code works.

I've given a try to the same code using 2 MBit data rate and I can confirm V1 to V2 doesn't work:
- ❌ V1 (ch0) -> V2 (ch0)
- ✅ V2 (ch0) -> V1(ch0)

If I then change the V1 channel from `0` to channel `1`, I can send data from V1 to V2, but not the other way:
- ✅  V1 (ch1) -> V2 (ch0)
- ❌ V2 (ch0) -> V1(ch1)

Is this the same issue you were encountering? Can you still replicate these results?

segrovets, 2022-08-09:
This is the same issue I was encountering
I can confirm I am reproducing the following behaviour :
> I've given a try to the same code using 2 MBit data rate and I can confirm V1 to V2 doesn't work:
> 
>     * x V1 (ch0) -> V2 (ch0)
> 
>     * white_check_mark V2 (ch0) -> V1(ch0)

Likewise this is also reproducible on my system.
> If I then change the V1 channel from `0` to channel `1`, I can send data from V1 to V2, but not the other way:
> 
>     * white_check_mark  V1 (ch1) -> V2 (ch0)
> 
>     * x V2 (ch0) -> V1(ch1)

Thank you for your follow up.

rhubarbdog, 2022-10-26:
hi i've got 2 version one and 2 version two microbits and would like to reproduce this.
what version of micropython are you running?