bbcmicrobit/micropython #773: where is pin 16 data on version 2 microbits

rhubarbdog, 2022-10-01:
i have written a DHT11 thermometer class for version 2 microbits.  it contains a dictionary of pin to bit id for example pin 0 uses bit 2.
I use a program in assembly to grab the memory at 0x5000510 and return it as an integer and print the binary with this function
```
def binary(value):

    if value<0:
	print(1,end=""
        value=value+(1<<31)
    else:
        print(0,end="")

    check=1<<30
    while check>0:
        if value>=check:
            print(1, end="")
            value=value-check
	else:
            print(0, end="")

        check = check // 2
```

when i try to find the bit that corresponds to pin16 nothing. i can only conclude that the bit for  pin16 is in another machine word because  `print(pin16.read_digital())` yeilds the correct result

microbit-carlos, 2022-10-11:
The pinout can be found in: https://tech.microbit.org/hardware/edgeconnector/#pins-and-signals

P16 in the edge connector is P0.16 in V2, so it's not in a different port (pins starting with `P1.xx`).

microbit-carlos, 2022-10-11:
My bad, there is an issue in the tech side and the V1 and V2 tables are reversed:
- https://github.com/microbit-foundation/dev-docs/issues/357

So pin16 is in fact in port 1, as it is the chip pin P1.02.

microbit-carlos, 2022-10-19:
Btw, looking at the datasheet for the nRF52833 (the MCU in the micro:bit V2), the `IN` register for pin port `P0` is at address 0x5000_0510 (base address 0x5000_0000 + 0x510 offset), and for port `P1` should be  0x5000_0810 (base address 0x5000_0300 + 0x510 offset).

[image]

And as the https://tech.microbit.org/hardware/edgeconnector/#pins-and-signals table shows, there are a few pins on port P1.

I'll close this as resolved, but feel free to ask any other questions you might have, just make sure to open the issues specific to micro:bit V2 in https://github.com/microbit-foundation/micropython-microbit-v2/.

Thanks!

closed 2022-10-19 by microbit-carlos