bbcmicrobit/micropython #774: degree symbol missing from version 2

rhubarbdog, 2022-10-01:
when i print a temperature or angle i like to use the degrees symbol °, i use `u'\x0b'` which works with version 1 microbits.

whilst on version 2 micobits `print("21%sC", u'\xb0')` yeilds 21�c

microbit-carlos, 2022-10-19:
Thanks for the report @rhubarbdog!

I think the first to cover is that `0x0B` is the value for `°` in (extended) ASCII, and the UTF-8 value is `0xC2B0`:
- https://www.rapidtables.com/code/text/ascii-table.html
- https://unicode-table.com/en/00B0/

Using this script below I've checked the output in the micro:bit Python editor, with the WebUSB serial terminal and seems to work more or less well: https://python.microbit.org

```python
import microbit

def is_v2():
    return hasattr(microbit, "microphone")

print("ASCII version:")
print("21%sC" % u"\xb0")
print("22{}C".format(u"\xb0"))
print("23\xb0C")
if is_v2():
    # V1 doesn't have the decode method in the bytes class
    print(b"24\xb0C".decode("ASCII"))

print()

print("UTF-8 version:")
print("25{}C".format(u"\xc2\xb0"))
print("26\xc2\xb0C")
if is_v2():
    # V1 doesn't have the decode method in the bytes class
    print(b"27\xc2\xb0C".decode("utf-8"))
print("28°C")
```
| Output V1 | Outpu V2 |
|----|----|
| [image] | [image] |

On the other hand, using a serial terminal in macOS, that let's me set the text encoding shows this:

| V1 terminal ASCII | V1 terminal UTF-8 |
|----|----|
| [image] | [image] |

| V2 terminal ASCII | V2 terminal UTF-8 |
|----|----|
 | [image] | [image] |

dpgeorge, 2022-10-24:
This should be fixed now that unicode is enabled on v2.

microbit-carlos, 2022-10-24:
Thanks Damien!

@rhubarbdog I'll close this as resolved in the V2 MicroPython codebase, but it will take a bit of time before we update MicroPython in the Python Editor.

If you have been using MicroPython with local tools like microFS you can download and test the latest CI build:
https://github.com/microbit-foundation/micropython-microbit-v2/actions/runs/3312585708

closed 2022-10-24 by microbit-carlos