bbcmicrobit/micropython #782: what about micoREPL?

rhubarbdog, 2022-10-25:
shouldn't instructions for `microrepl` be included on this page https://microbit-micropython.readthedocs.io/en/v2-docs/devguide/repl.html

microbit-carlos, 2022-11-07:
Yes, we can add that to the https://microbit-micropython.readthedocs.io/en/v2-docs/devguide/repl.html#using-a-serial-communication-program section, between the first and second paragraph would be good.
Would you like to create a PR to add it?

rhubarbdog, 2022-11-07:
Yes I can do

rhubarbdog, 2022-11-08:
i've completed the PR its https://github.com/bbcmicrobit/micropython/pull/785 if you want to check it's ok

microbit-carlos, 2022-11-18:
PR https://github.com/bbcmicrobit/micropython/pull/785 was merged and this can be closed, thanks @rhubarbdog!

closed 2022-11-18 by microbit-carlos