bbcmicrobit/micropython #783: microFS and REPL

rhubarbdog, 2022-10-25:
under linux if a serial device such as screen is still attached to the micobit because one inadvertently detached the screen rather than quit then `ufs` give this error under linux
`[Errno 16] could not open port /dev/ttyACM0` there will be similar errors under macOS and windows.
is this worth mentioning on this page https://microbit-micropython.readthedocs.io/en/v2-docs/devguide/repl.html  in readthedocs

microbit-carlos, 2022-11-07:
I think that's already covered in this paragraph, no?
> There are many ways back to a command prompt including Ctrl-A then Ctrl-D, which will detach screen. All serial output from the micro:bit will still be received by screen, the serial port will be locked, preventing other applications from accessing it. You can restart screen by typing screen -r.

rhubarbdog, 2022-11-07:
Yes it does

closed 2022-11-07 by rhubarbdog