bbcmicrobit/micropython #784: MicroPython consumes too much memory and causes memory allocation error on V1 micro:bits

JackAtKitronik, 2022-11-04:
**Describe the bug**
Using the micro:bit Python Editor, we were trying to add servo and LED functionality to the Kitronik MOVE Motor MicroPython library. Upon adding the servo code (about 20 lines of code) the code is now too big and when put on the V1 micro:bit it causes a memory allocation error saying memeory allocation failed, allocating 398 bytes.

**To Reproduce**
Steps to reproduce the behavior:
1. Open https://python.microbit.org/v/3/project
2. Save [KitronikMOVEMotor.py](https://github.com/DaveAtKitronik/micropython-microbit-kitronik-MOVE-motor/blob/master/KitronikMOVEMotor.py)
3. Add the code at [MOVEMotor.py](https://github.com/DaveAtKitronik/micropython-microbit-kitronik-MOVE-motor/blob/master/MOVEMotor.py) to main.py
4. Download the code onto a V1 micro:bit
5. The error `memory error, memeory allocation failed, allocating 398 bytes` appears

**Expected behavior**
Following the same steps on as above but the program works on a V1 micro:bit.
On a V2 micro:bit then the code works as expected.

**micro:bit version**
V1 micro:bit

dpgeorge, 2022-11-07:
I don't think there's much that can be done about this, the micro:bit V1 just doesn't have enough memory.

If you really need to get it working on a V1, it might be possible to optimise the code so that it does fit.  But in general that's not possible to do, and the way to resolve it is either:
1. have simpler programs
2. use a micro:bit V2

microbit-carlos, 2022-11-07:
@JackAtKitronik was this the issue where the code worked fine in a single file but it was running out of memory (in this case by ~400 bytes) when using a module?

@dpgeorge if that's the case, would that be expected? that separating the code into a module could consume that much extra memory?

rhubarbdog, 2022-11-07:
I got this working and used it with a move:mini https://github.com/rhubarbdog/microbit-servo

dpgeorge, 2022-11-07:
> if that's the case, would that be expected? that separating the code into a module could consume that much extra memory?

Yes that's expected, simply because you now have two .py files: the main one which imports the second one.  And some memory is already used by the first at the point where the second is imported and compiled.

This case would definitely benefit from being able to import .mpy files.

JackAtKitronik, 2022-11-08:
> @JackAtKitronik was this the issue where the code worked fine in a single file but it was running out of memory (in this case by ~400 bytes) when using a module?

@microbit-carlos almost. Putting it into a single file did make it smaller but not small enough to work.

> I got this working and used it with a move:mini https://github.com/rhubarbdog/microbit-servo

@rhubarbdog great I'll have a look.