bbcmicrobit/micropython #788: Funky Display issue

rhubarbdog, 2022-12-19:
when i run program
```
from microbit import *

display.scroll("Game over", wait = False, loop = True)

while True:
    for i in range(5):
        display.set_pixel(i, i, 1)
        display.set_pixel(4 - i, i, 1)

    sleep(100)
```
i get both the cross displayed and the scrolling message "Game over"

microbit-carlos, 2022-12-19:
Yes, I think this works as expected, as the `set_pixel` will modify the current display image, while the scroll animation will overwrite the image being displayed. So it will look like the X is drawn on top of the text, but it will flicker (and the flicker should be reduced by removing the sleep in the while loop).

What was your expectation for this programme to run?

rhubarbdog, 2022-12-19:
I'm not sure.
I knew i had a bug in my code when i experienced this and knew what the issue was.
i only reported it as a potential bug, it's one of a few solutions when mixing `display.show`, `display.scroll` and `display.set_pixel`

microbit-carlos, 2022-12-23:
Hopefully I've been able to clarify the behaviour, so I'll close this as resolved, but do let me know if you do have any other questions.

closed 2022-12-23 by microbit-carlos