bbcmicrobit/micropython #789: display.clear() does not stop looping animations

davidtkaye, 2023-02-16:
If a display.show or display.scroll function is called with the loop parameter set to True and the wait parameter set to False then an infinite loop of text/animation is shown on the display. Any subsequent show or scroll function call will break this loop, however the display.clear function will not do this. Therefore to stop an animation and clear the screen two function calls are needed (or a custom image of all 0s).

This workaround is fine but not totally intuitive: the behaviour of the display.clear function should ideally stop any running animation, or if not there should be a straightforward option to do this.

microbit-carlos, 2023-03-17:
Thanks for the report @davidtkaye!

I tried this programme with a BBC micro:bit V1 and V2 and I think the issue is actually only in V2. Which version of the board did you use?

```python
from microbit import *

display.show("hello", loop=True, wait=False)
while True:
    sleep(100)
    if button_a.is_pressed():
        display.clear()
    if button_b.is_pressed():
        display.show("X")
```

davidtkaye, 2023-03-18:
Yep, sorry. Should have been in the v2 repository!