bbcmicrobit/micropython #798: There isn't a SCISSORS image

rhubarbdog, 2024-01-04:
`display.show(Image.SCISSORS)` causes error
`AttributeError: type object 'MicroBitImage' has no attribute 'SCISSORS'`

readthedocs lists scissors as a built in image

dpgeorge, 2024-01-10:
The SCISSORS image was added quite a while ago in commit 5005880aa3acd5e1fde9b09b9de85932e65353f1, and it works for me on a micro:bit v1 with firmware v1.1.1.  And also a micro:bit v2 with firmware v2.1.1.

rhubarbdog, 2024-01-16:
uflash claims it's up to date. is there a microbit.hex that is at 2.1.1 or do i need to compile one

dpgeorge, 2024-01-18:
If you use the online Python editor then it should be able to update your micro:bit to the latest firmware: https://python.microbit.org/v/3

rhubarbdog, 2024-01-18:
the editor gives me version 2.1.2 which contains scissors